Farama-Foundation/Gymnasium-Robotics #5: [Bug Report] error with 'pip install gym-robotics'

perrin-isir, 2022-02-21:
**Describe the bug**
After `pip install gym` (version 0.22.0), `pip install gym-robotics` leads to the following error:
```
Packages installed from PyPI cannot depend on packages which are not also hosted on PyPI.
```

And when installing directly gym-robotics with pip (without a prior installation of gym): `pip install gym-robotics`, the import of gym_robotics returns the following error:
``` 
File "<stdin>", line 1, in <module>
  File "/home/perrin/.conda/envs/p3/lib/python3.7/site-packages/gym_robotics/__init__.py", line 1, in <module>
    from gym.envs.registration import register
  File "/home/perrin/.conda/envs/p3/lib/python3.7/site-packages/gym/__init__.py", line 12, in <module>
    from gym.envs import make, spec, register
  File "/home/perrin/.conda/envs/p3/lib/python3.7/site-packages/gym/envs/__init__.py", line 10, in <module>
    _load_env_plugins()
  File "/home/perrin/.conda/envs/p3/lib/python3.7/site-packages/gym/envs/registration.py", line 725, in load_env_plugins
    fn = plugin.load()
  File "/home/perrin/.conda/envs/p3/lib/python3.7/site-packages/importlib_metadata/__init__.py", line 196, in load
    return functools.reduce(getattr, attrs, module)
AttributeError: module 'gym_robotics' has no attribute 'register_robotics_envs'
```

**System Info**
Ubuntu 22.04
Tested with Python 3.7.7 and 3.9.9

nikhilweee, 2022-02-25:
Here's what I get:
```
$ pip install gym-robotics
Collecting gym-robotics
  Downloading gym-robotics-0.0.2.tar.gz (1.5 MB)
     |████████████████████████████████| 1.5 MB 35.9 MB/s 
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
ERROR: Packages installed from PyPI cannot depend on packages which are not also hosted on PyPI.
gym-robotics depends on gym@ git+https://github.com/openai/gym@master 
```

jkterry1, 2022-02-25:
Hey, we're aware of this. @vwxyzjn should be working on a fix soon.

vwxyzjn, 2022-02-25:
It's done. See https://github.com/Farama-Foundation/gym-robotics/releases/tag/v0.1.0

closed 2022-02-25 by vwxyzjn