Farama-Foundation/Gymnasium-Robotics #11: [Question] Observations meaning

Borzyszkowski, 2022-03-07:
Hello!

I would like to ask what do the observations in environments mean specifically? I'm especially interested in 'HandReach-v0' and there are 63 observables, however I could not find and information about their meaning.

For example, the observations for OpenAI Gym are described here:
https://github.com/openai/gym/wiki/CartPole-v0

jkterry1, 2022-03-08:
The proper documentation for cartpole is here: https://www.gymlibrary.ml/pages/environments/classic_control/cart_pole

Theese environments don't have any equivalent documentation unfortunately. We'd be happy to take PRs adding it though.

Borzyszkowski, 2022-03-11:
It is quite surprising that there is no documentation for the model at all. Do you know if there are any other sources that could explain how the sensors work or how to interpret the hand position? I've tried to get it out of [the code](
https://github.com/Farama-Foundation/Gym-Robotics/blob/main/gym_robotics/envs/hand/manipulate_touch_sensors.py#L117), but it's very vague.

jkterry1, 2022-03-12:
Not really, beyond what's in the env code. None of the environments in Gym were documented until I took over them. Documenting the observation space of these would be an enormous amount of work, and we haven't done it yet, and as these aren't in Gym proper anymore it hasn't been at the top of our to-do list either.  I'd be very happy to merge .md files documenting this though

qgallouedec, 2022-03-12:
For the Fetch environments

- gripper position (3 coord.): (x, y, z) of the gripper.
- object position (3 coord.): (x, y, z) of the object. (0, 0, 0) if no object.
- object relative position (3 coord.) : gripper position - object position
- gripper state (2 coord.): (w/2, w/2) the two coordinates are egal, and the sum of them is the distance between the fingers.
- object rotation (3 coord.): (rx, ry, rz) ; (0, 0, 0) if no object.
- object velocity (3 coord.): (vx, vy, vz) ; (0, 0, 0) if no object.
- object rotationnal velocity (3 coord.): (rvx, rvy, rvz) ; (0, 0, 0) if no object.
- gripper velocity (3 coord.): (vx, vy, vz)
- fingers velocity, (2 coord.): (dw/2, dw/2)

In total, there are 25 coordinates for all fetch environments.

jkterry1, 2022-03-12:
@qgallouedec that's super helpful- would you be willing to contribute .md files in a docs folder describing the 4 fetch environments?

qgallouedec, 2022-03-12:
I'll do that

Borzyszkowski, 2022-03-21:
Thank you so much - this is great.
I'm closing the topic as it seems to be resolved.

closed 2022-03-21 by Borzyszkowski