Farama-Foundation/Gymnasium-Robotics #14: [Proposal] Use mujoco bindings from deepmind instead of mujoco_py

arjun-kg, 2022-05-25:
Since `mujoco_py` is no longer maintained, and gym has also already updated its other mujoco envs to use deepmind's python bindings, `gym-robotics` needs to be updated as well.

wookayin, 2022-08-20:
Hello is there any update on this?  Due to the current dependency to `mujoco_py`, gym-robotics environments (e.g., `FetchSlide-v1`) cannot be used when a recent setup is used (mujoco 2.2+, gym 0.25.1, etc.). Looking forward to it, thanks!

rodrigodelazcano, 2022-08-20:
Hi @wookayin, thanks for your interest. A PR will be made soon with new versions of the environments using the latest `mujoco` bindings and compatibility with `gym` API. If you want to follow the work the forked repo I'm working on is this one: https://github.com/rodrigodelazcano/Gym-Robotics/tree/mujoco-bindings.

Currently I'm performing some training benchmarks of the new environment versions against the old ones.

rodrigodelazcano, 2022-09-14:
@wookayin @arjun-kg the latest environment versions have the new mujoco bindings. This PR has also the benchmarks #18

closed 2022-09-14 by rodrigodelazcano