Farama-Foundation/Gymnasium-Robotics #19: [Bug Report] Minor User Warning : WARN: Overriding environment HandManipulateBlockRotateZ-v1 already in registry.

leonasting, 2022-09-20:
**Describe the bug**
Getting user warning  UserWarning: WARN: Overriding environment HandManipulateBlockRotateZ-v1 already in registry.
when importing Gym.
**Code example**

**System Info**
Describe the characteristic of your environment:
gym 0.26
python 3.10
windows
 * Python version
### Checklist

[image]
Updating the init file will fix it.

rodrigodelazcano, 2022-09-22:
Thank you @leonasting for notifying. Will you be able to make a PR with the changes?

leonasting, 2022-09-22:
Yes, I have created a request. Pull https://github.com/Farama-Foundation/Gym-Robotics/pull/20

closed 2022-09-25 by rodrigodelazcano