Farama-Foundation/Gymnasium-Robotics #55: [Proposal] MinariDataset Generation

WillDudley, 2022-12-11:
### Proposal 

Up-to-date, re-runnable expert Minari datasets for offline learning.

### Motivation

Minari is now at a stage where datasets can be made and uploaded for experimental use by others.

### Pitch

- Have an expert policy perform 100k rollouts on each environment, save this buffer as a MinariDataset and upload it for public use.

### Alternatives

If anyone thinks Minari needs to be improved before doing this, let me know.

### Additional context

See https://github.com/Farama-Foundation/Minari/blob/main/tutorials/DatasetCreation/dataset_creation.py for an example of creating and uploading a MinariDataset.

### Checklist

- [ ] I have checked that there is no similar [issue](https://github.com/Farama-Foundation/gymnasium-robotics/issues) in the repo (**required**)

Kallinteris-Andreas, 2024-01-08:
Update from jan-2024: [Shadow Dexterous Hand](https://robotics.farama.org/envs/shadow_dexterous_hand/) & [Fetch](https://robotics.farama.org/envs/fetch/) are the only missing environments

jkterry1, 2026-10-09:
Are those environments still missing?