Farama-Foundation/Gymnasium-Robotics #56: [Bug Report] Doc Generation gives warnings

Kallinteris-Andreas, 2022-12-13:
```
[~/Gymnasium-Robotics]$ cd docs              
[~/Gymnasium-Robotics/docs]$ make dirhtml _build
Running Sphinx v5.2.3
loading pickled environment... done
myst v0.18.1: MdParserConfig(commonmark_only=False, gfm_only=False, enable_extensions=[], disable_syntax=[], all_links_external=False, url_schemes=('http', 'https', 'mailto', 'ftp'), ref_domains=None, highlight_code_blocks=True, number_code_blocks=[], title_to_header=False, heading_anchors=None, heading_slug_func=None, footnote_transition=True, words_per_minute=200, sub_delimiters=('{', '}'), linkify_fuzzy_links=True, dmath_allow_labels=True, dmath_allow_space=True, dmath_allow_digits=True, dmath_double_inline=False, update_mathjax=True, mathjax_classes='tex2jax_process|mathjax_process|math|output_area')
building [mo]: targets for 0 po files that are out of date
building [dirhtml]: targets for 0 source files that are out of date
updating environment: 0 added, 3 changed, 0 removed
reading sources... [100%] envs/hand_touch/index                                  
/home/master-andreas/Gymnasium-Robotics/docs/envs/fetch/index.md:6: WARNING: Document headings start at H2, not H1 [myst.header]
/home/master-andreas/Gymnasium-Robotics/docs/envs/fetch/index.md:15: CRITICAL: Problems with "raw" directive path:
InputError: [Errno 2] No such file or directory: 'envs/fetch/list.html'.
/home/master-andreas/Gymnasium-Robotics/docs/envs/fetch/index.md:19: WARNING: toctree contains reference to nonexisting document 'envs/fetch/FetchReach'
/home/master-andreas/Gymnasium-Robotics/docs/envs/fetch/index.md:19: WARNING: toctree contains reference to nonexisting document 'envs/fetch/FetchSlide'
/home/master-andreas/Gymnasium-Robotics/docs/envs/fetch/index.md:19: WARNING: toctree contains reference to nonexisting document 'envs/fetch/FetchPickAndPlace'
/home/master-andreas/Gymnasium-Robotics/docs/envs/fetch/index.md:19: WARNING: toctree contains reference to nonexisting document 'envs/fetch/FetchPush'
/home/master-andreas/Gymnasium-Robotics/docs/envs/hand/index.md:6: WARNING: Document headings start at H2, not H1 [myst.header]
/home/master-andreas/Gymnasium-Robotics/docs/envs/hand/index.md:15: CRITICAL: Problems with "raw" directive path:
InputError: [Errno 2] No such file or directory: 'envs/hand/list.html'.
/home/master-andreas/Gymnasium-Robotics/docs/envs/hand/index.md:19: WARNING: toctree contains reference to nonexisting document 'envs/hand/HandReach'
/home/master-andreas/Gymnasium-Robotics/docs/envs/hand/index.md:19: WARNING: toctree contains reference to nonexisting document 'envs/hand/HandBlock'
/home/master-andreas/Gymnasium-Robotics/docs/envs/hand/index.md:19: WARNING: toctree contains reference to nonexisting document 'envs/hand/HandEgg'
/home/master-andreas/Gymnasium-Robotics/docs/envs/hand/index.md:19: WARNING: toctree contains reference to nonexisting document 'envs/hand/HandPen'
/home/master-andreas/Gymnasium-Robotics/docs/envs/hand_touch/index.md:20: CRITICAL: Problems with "raw" directive path:
InputError: [Errno 2] No such file or directory: 'envs/hand_touch/list.html'.
/home/master-andreas/Gymnasium-Robotics/docs/envs/hand_touch/index.md:24: WARNING: toctree contains reference to nonexisting document 'envs/hand_touch/HandBlockTouchSensors'
/home/master-andreas/Gymnasium-Robotics/docs/envs/hand_touch/index.md:24: WARNING: toctree contains reference to nonexisting document 'envs/hand_touch/HandEggTouchSensors'
/home/master-andreas/Gymnasium-Robotics/docs/envs/hand_touch/index.md:24: WARNING: toctree contains reference to nonexisting document 'envs/hand_touch/HandPenTouchSensors'
looking for now-outdated files... none found
pickling environment... done
checking consistency... /home/master-andreas/Gymnasium-Robotics/docs/404.md: WARNING: document isn't included in any toctree
/home/master-andreas/Gymnasium-Robotics/docs/README.md: WARNING: document isn't included in any toctree
done
preparing documents... done
writing output... [100%] index                                                   
generating indices... genindex done
writing additional pages... search done
copying images... [100%] _static/videos/fetch/FetchPickAndPlace.gif              
copying static files... done
copying extra files... done
dumping search index in English (code: en)... done
dumping object inventory... done
build succeeded, 18 warnings.

The HTML pages are in _build/dirhtml.
Running Sphinx v5.2.3

Sphinx error:
Builder name _build not registered or available through entry point
make: *** [Makefile:20: _build] Error 2

```

mgoulao, 2022-12-19:
Just run `make dirhtml`

Kallinteris-Andreas, 2022-12-20:
@mgoulao  same thing

mgoulao, 2022-12-20:
Ok, can you try this: `sphinx-build -b dirhtml -v . _build`

Kallinteris-Andreas, 2022-12-20:
```
$ sphinx-build -b dirhtml -v . _build
Running Sphinx v5.2.3
locale_dir /home/master-andreas/ai/Gymnasium-Robotics-Kalli/docs/locales/en/LC_MESSAGES does not exists
myst v0.18.1: MdParserConfig(commonmark_only=False, gfm_only=False, enable_extensions=[], disable_syntax=[], all_links_external=False, url_schemes=('http', 'https', 'mailto', 'ftp'), ref_domains=None, highlight_code_blocks=True, number_code_blocks=[], title_to_header=False, heading_anchors=None, heading_slug_func=None, footnote_transition=True, words_per_minute=200, sub_delimiters=('{', '}'), linkify_fuzzy_links=True, dmath_allow_labels=True, dmath_allow_space=True, dmath_allow_digits=True, dmath_double_inline=False, update_mathjax=True, mathjax_classes='tex2jax_process|mathjax_process|math|output_area')
locale_dir /home/master-andreas/ai/Gymnasium-Robotics-Kalli/docs/locales/en/LC_MESSAGES does not exists
building [mo]: targets for 0 po files that are out of date
building [dirhtml]: targets for 10 source files that are out of date
updating environment: locale_dir /home/master-andreas/ai/Gymnasium-Robotics-Kalli/docs/locales/en/LC_MESSAGES does not exists
[new config] 10 added, 0 changed, 0 removed
reading sources... [ 10%] 404
reading sources... [ 20%] README
reading sources... [ 30%] content/installation
reading sources... [ 40%] content/multi-goal_api
reading sources... [ 50%] envs/MaMuJoCo/index
reading sources... [ 60%] envs/MaMuJoCo/ma_ant
reading sources... [ 70%] envs/fetch/index
reading sources... [ 80%] envs/hand/index
reading sources... [ 90%] envs/hand_touch/index
reading sources... [100%] index

/home/master-andreas/ai/Gymnasium-Robotics-Kalli/docs/content/installation.md:6: WARNING: Document headings start at H2, not H1 [myst.header]
/home/master-andreas/ai/Gymnasium-Robotics-Kalli/docs/content/multi-goal_api.md:7: WARNING: Document headings start at H2, not H1 [myst.header]
/home/master-andreas/ai/Gymnasium-Robotics-Kalli/docs/envs/MaMuJoCo/index.md:7: WARNING: Document headings start at H2, not H1 [myst.header]
/home/master-andreas/ai/Gymnasium-Robotics-Kalli/docs/envs/MaMuJoCo/index.md:27: CRITICAL: Problems with "raw" directive path:
InputError: [Errno 2] No such file or directory: 'envs/MaMuJoCo/list.html'.
/home/master-andreas/ai/Gymnasium-Robotics-Kalli/docs/envs/fetch/index.md:6: WARNING: Document headings start at H2, not H1 [myst.header]
/home/master-andreas/ai/Gymnasium-Robotics-Kalli/docs/envs/fetch/index.md:15: CRITICAL: Problems with "raw" directive path:
InputError: [Errno 2] No such file or directory: 'envs/fetch/list.html'.
/home/master-andreas/ai/Gymnasium-Robotics-Kalli/docs/envs/fetch/index.md:19: WARNING: toctree contains reference to nonexisting document 'envs/fetch/FetchReach'
/home/master-andreas/ai/Gymnasium-Robotics-Kalli/docs/envs/fetch/index.md:19: WARNING: toctree contains reference to nonexisting document 'envs/fetch/FetchSlide'
/home/master-andreas/ai/Gymnasium-Robotics-Kalli/docs/envs/fetch/index.md:19: WARNING: toctree contains reference to nonexisting document 'envs/fetch/FetchPickAndPlace'
/home/master-andreas/ai/Gymnasium-Robotics-Kalli/docs/envs/fetch/index.md:19: WARNING: toctree contains reference to nonexisting document 'envs/fetch/FetchPush'
/home/master-andreas/ai/Gymnasium-Robotics-Kalli/docs/envs/hand/index.md:6: WARNING: Document headings start at H2, not H1 [myst.header]
/home/master-andreas/ai/Gymnasium-Robotics-Kalli/docs/envs/hand/index.md:15: CRITICAL: Problems with "raw" directive path:
InputError: [Errno 2] No such file or directory: 'envs/hand/list.html'.
/home/master-andreas/ai/Gymnasium-Robotics-Kalli/docs/envs/hand/index.md:19: WARNING: toctree contains reference to nonexisting document 'envs/hand/HandReach'
/home/master-andreas/ai/Gymnasium-Robotics-Kalli/docs/envs/hand/index.md:19: WARNING: toctree contains reference to nonexisting document 'envs/hand/HandBlock'
/home/master-andreas/ai/Gymnasium-Robotics-Kalli/docs/envs/hand/index.md:19: WARNING: toctree contains reference to nonexisting document 'envs/hand/HandEgg'
/home/master-andreas/ai/Gymnasium-Robotics-Kalli/docs/envs/hand/index.md:19: WARNING: toctree contains reference to nonexisting document 'envs/hand/HandPen'
/home/master-andreas/ai/Gymnasium-Robotics-Kalli/docs/envs/hand_touch/index.md:20: CRITICAL: Problems with "raw" directive path:
InputError: [Errno 2] No such file or directory: 'envs/hand_touch/list.html'.
/home/master-andreas/ai/Gymnasium-Robotics-Kalli/docs/envs/hand_touch/index.md:24: WARNING: toctree contains reference to nonexisting document 'envs/hand_touch/HandBlockTouchSensors'
/home/master-andreas/ai/Gymnasium-Robotics-Kalli/docs/envs/hand_touch/index.md:24: WARNING: toctree contains reference to nonexisting document 'envs/hand_touch/HandEggTouchSensors'
/home/master-andreas/ai/Gymnasium-Robotics-Kalli/docs/envs/hand_touch/index.md:24: WARNING: toctree contains reference to nonexisting document 'envs/hand_touch/HandPenTouchSensors'
looking for now-outdated files... none found
pickling environment... done
checking consistency... /home/master-andreas/ai/Gymnasium-Robotics-Kalli/docs/404.md: WARNING: document isn't included in any toctree
/home/master-andreas/ai/Gymnasium-Robotics-Kalli/docs/README.md: WARNING: document isn't included in any toctree
done
preparing documents... done
writing output... [ 10%] 404
writing output... [ 20%] README
writing output... [ 30%] content/installation
writing output... [ 40%] content/multi-goal_api
writing output... [ 50%] envs/MaMuJoCo/index
[... cut ...]
writing output... [100%] index

/home/master-andreas/ai/Gymnasium-Robotics-Kalli/docs/README.md:4: WARNING: 'myst' reference target not found: 
generating indices... genindex done
writing additional pages... search done
copying images... [100%] _static/videos/fetch/FetchPickAndPlace.gif

copying static files... done
copying extra files... done
dumping search index in English (code: en)... done
dumping object inventory... done
build succeeded, 23 warnings.

The HTML pages are in _build.
```

same issue

mgoulao, 2022-12-20:
What is the issue? The `InputError: [Errno 2] No such file or directory: 'envs/hand_touch/list.html'` is normal if you haven't run the [gen_envs_display.py](https://github.com/Farama-Foundation/Gymnasium-Robotics/blob/main/docs/scripts/gen_envs_display.py) and is not needed when writing the docs.

Kallinteris-Andreas, 2022-12-20:
1) this does not remove all the warnings
```sh
[master-andreas@master-pc ~/ai/Gymnasium-Robotics-Kalli/docs]$ py scripts/gen_envs_display.py
[master-andreas@master-pc ~/ai/Gymnasium-Robotics-Kalli/docs]$ make dirhtml                  
Running Sphinx v5.2.3
making output directory... done
myst v0.18.1: MdParserConfig(commonmark_only=False, gfm_only=False, enable_extensions=[], disable_syntax=[], all_links_external=False, url_schemes=('http', 'https', 'mailto', 'ftp'), ref_domains=None, highlight_code_blocks=True, number_code_blocks=[], title_to_header=False, heading_anchors=None, heading_slug_func=None, footnote_transition=True, words_per_minute=200, sub_delimiters=('{', '}'), linkify_fuzzy_links=True, dmath_allow_labels=True, dmath_allow_space=True, dmath_allow_digits=True, dmath_double_inline=False, update_mathjax=True, mathjax_classes='tex2jax_process|mathjax_process|math|output_area')
building [mo]: targets for 0 po files that are out of date
building [dirhtml]: targets for 10 source files that are out of date
updating environment: [new config] 10 added, 0 changed, 0 removed
reading sources... [100%] index                                                                                                                                                                                                               
/home/master-andreas/ai/Gymnasium-Robotics-Kalli/docs/content/installation.md:6: WARNING: Document headings start at H2, not H1 [myst.header]
/home/master-andreas/ai/Gymnasium-Robotics-Kalli/docs/content/multi-goal_api.md:7: WARNING: Document headings start at H2, not H1 [myst.header]
/home/master-andreas/ai/Gymnasium-Robotics-Kalli/docs/envs/MaMuJoCo/index.md:7: WARNING: Document headings start at H2, not H1 [myst.header]
/home/master-andreas/ai/Gymnasium-Robotics-Kalli/docs/envs/MaMuJoCo/index.md:27: CRITICAL: Problems with "raw" directive path:
InputError: [Errno 2] No such file or directory: 'envs/MaMuJoCo/list.html'.
/home/master-andreas/ai/Gymnasium-Robotics-Kalli/docs/envs/fetch/index.md:6: WARNING: Document headings start at H2, not H1 [myst.header]
/home/master-andreas/ai/Gymnasium-Robotics-Kalli/docs/envs/fetch/index.md:19: WARNING: toctree glob pattern './*' didn't match any documents
/home/master-andreas/ai/Gymnasium-Robotics-Kalli/docs/envs/hand/index.md:6: WARNING: Document headings start at H2, not H1 [myst.header]
/home/master-andreas/ai/Gymnasium-Robotics-Kalli/docs/envs/hand/index.md:19: WARNING: toctree glob pattern './*' didn't match any documents
/home/master-andreas/ai/Gymnasium-Robotics-Kalli/docs/envs/hand_touch/index.md:24: WARNING: toctree glob pattern './*' didn't match any documents
looking for now-outdated files... none found
pickling environment... done
checking consistency... /home/master-andreas/ai/Gymnasium-Robotics-Kalli/docs/404.md: WARNING: document isn't included in any toctree
/home/master-andreas/ai/Gymnasium-Robotics-Kalli/docs/README.md: WARNING: document isn't included in any toctree
done
preparing documents... done
writing output... [100%] index                                                                                                                                                                                                                
/home/master-andreas/ai/Gymnasium-Robotics-Kalli/docs/README.md:4: WARNING: 'myst' reference target not found: 
generating indices... genindex done
writing additional pages... search done
copying images... [100%] _static/videos/fetch/FetchPickAndPlace.gif                                                                                                                                                                           
copying static files... done
copying extra files... done
dumping search index in English (code: en)... done
dumping object inventory... done
build succeeded, 12 warnings.

The HTML pages are in _build/dirhtml.
```
Here is an image (which includes colors)
[image]

2) This matters because the doc building guide: 
https://github.com/Farama-Foundation/Gymnasium/blob/main/docs/README.md#build-the-documentation
only says to run `make dirhtml _build`, and if someone gets warning, is likely to not contribute

rodrigodelazcano, 2023-01-10:
@mgoulao @Kallinteris-Andreas  what's the status on this?

closed 2023-01-13 by Kallinteris-Andreas