Farama-Foundation/Gymnasium-Robotics #65: [Bug Report] double registration import issue

Kallinteris-Andreas, 2022-12-25:
There are issues with importing `gymnasium_robotics/__init__.py` twice

```sh
$ pip list | grep gym           
gymnasium                     0.27.0
```

# Git version
```sh
~/Gymnasium-Robotics]$ git pull           
Already up to date
```
```python
~/Gymnasium-Robotics]$ pip install . && py
Defaulting to user installation because normal site-packages is not writeable
Processing /home/master-andreas/Gymnasium-Robotics
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... done
Requirement already satisfied: numpy<1.24.0,>=1.21.0 in /usr/lib/python3.10/site-packages (from gymnasium-robotics==1.1.0) (1.23.5)
Requirement already satisfied: mujoco>=2.3.1.post1 in /home/master-andreas/.local/lib/python3/site-packages (from gymnasium-robotics==1.1.0) (2.3.1.post1)
Requirement already satisfied: gymnasium>=0.26 in /home/master-andreas/.local/lib/python3/site-packages (from gymnasium-robotics==1.1.0) (0.27.0)
Requirement already satisfied: jax-jumpy>=0.2.0 in /home/master-andreas/.local/lib/python3/site-packages (from gymnasium>=0.26->gymnasium-robotics==1.1.0) (0.2.0)
Requirement already satisfied: typing-extensions>=4.3.0 in /usr/lib/python3.10/site-packages (from gymnasium>=0.26->gymnasium-robotics==1.1.0) (4.4.0)
Requirement already satisfied: shimmy<1.0,>=0.1.0 in /home/master-andreas/.local/lib/python3/site-packages (from gymnasium>=0.26->gymnasium-robotics==1.1.0) (0.2.0)
Requirement already satisfied: cloudpickle>=1.2.0 in /home/master-andreas/.local/lib/python3/site-packages (from gymnasium>=0.26->gymnasium-robotics==1.1.0) (2.2.0)
Requirement already satisfied: gymnasium-notices>=0.0.1 in /home/master-andreas/.local/lib/python3/site-packages (from gymnasium>=0.26->gymnasium-robotics==1.1.0) (0.0.1)
Requirement already satisfied: glfw in /home/master-andreas/.local/lib/python3/site-packages (from mujoco>=2.3.1.post1->gymnasium-robotics==1.1.0) (2.5.5)
Requirement already satisfied: absl-py in /home/master-andreas/.local/lib/python3/site-packages (from mujoco>=2.3.1.post1->gymnasium-robotics==1.1.0) (1.3.0)
Requirement already satisfied: pyopengl in /home/master-andreas/.local/lib/python3/site-packages (from mujoco>=2.3.1.post1->gymnasium-robotics==1.1.0) (3.1.6)
Building wheels for collected packages: gymnasium-robotics
  Building wheel for gymnasium-robotics (pyproject.toml) ... done
  Created wheel for gymnasium-robotics: filename=gymnasium_robotics-1.1.0-py3-none-any.whl size=1485120 sha256=79a9063e315f99716b0f7ff51ba6ceb5e7d82534aea3f4b88936593645b42db1
  Stored in directory: /home/master-andreas/.cache/pip/wheels/fb/22/bd/db3af103b5ded132ca76abe0558b08910f496de990b605d1d8
Successfully built gymnasium-robotics
Installing collected packages: gymnasium-robotics
  Attempting uninstall: gymnasium-robotics
    Found existing installation: gymnasium-robotics 1.1.0
    Uninstalling gymnasium-robotics-1.1.0:
      Successfully uninstalled gymnasium-robotics-1.1.0
Successfully installed gymnasium-robotics-1.1.0
Python 3.10.8 (main, Nov  1 2022, 14:18:21) [GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import gymnasium_robotics
/home/master-andreas/.local/lib/python3.10/site-packages/gymnasium/envs/registration.py:520: UserWarning: WARN: Overriding environment FetchSlide-v1 already in registry.
  logger.warn(f"Overriding environment {new_spec.id} already in registry.")
/home/master-andreas/.local/lib/python3.10/site-packages/gymnasium/envs/registration.py:520: UserWarning: WARN: Overriding environment FetchSlide-v2 already in registry.
  logger.warn(f"Overriding environment {new_spec.id} already in registry.")
/home/master-andreas/.local/lib/python3.10/site-packages/gymnasium/envs/registration.py:520: UserWarning: WARN: Overriding environment FetchPickAndPlace-v1 already in registry.
  logger.warn(f"Overriding environment {new_spec.id} already in registry.")
/home/master-andreas/.local/lib/python3.10/site-packages/gymnasium/envs/registration.py:520: UserWarning: WARN: Overriding environment FetchPickAndPlace-v2 already in registry.
  logger.warn(f"Overriding environment {new_spec.id} already in registry.")
/home/master-andreas/.local/lib/python3.10/site-packages/gymnasium/envs/registration.py:520: UserWarning: WARN: Overriding environment FetchReach-v1 already in registry.
  logger.warn(f"Overriding environment {new_spec.id} already in registry.")
/home/master-andreas/.local/lib/python3.10/site-packages/gymnasium/envs/registration.py:520: UserWarning: WARN: Overriding environment FetchReach-v2 already in registry.
  logger.warn(f"Overriding environment {new_spec.id} already in registry.")
/home/master-andreas/.local/lib/python3.10/site-packages/gymnasium/envs/registration.py:520: UserWarning: WARN: Overriding environment FetchReach-v3 already in registry.
  logger.warn(f"Overriding environment {new_spec.id} already in registry.")
/home/master-andreas/.local/lib/python3.10/site-packages/gymnasium/envs/registration.py:520: UserWarning: WARN: Overriding environment FetchPush-v1 already in registry.
  logger.warn(f"Overriding environment {new_spec.id} already in registry.")
/home/master-andreas/.local/lib/python3.10/site-packages/gymnasium/envs/registration.py:520: UserWarning: WARN: Overriding environment FetchPush-v2 already in registry.
  logger.warn(f"Overriding environment {new_spec.id} already in registry.")
/home/master-andreas/.local/lib/python3.10/site-packages/gymnasium/envs/registration.py:520: UserWarning: WARN: Overriding environment HandReach-v0 already in registry.
  logger.warn(f"Overriding environment {new_spec.id} already in registry.")
/home/master-andreas/.local/lib/python3.10/site-packages/gymnasium/envs/registration.py:520: UserWarning: WARN: Overriding environment HandReach-v1 already in registry.
  logger.warn(f"Overriding environment {new_spec.id} already in registry.")
/home/master-andreas/.local/lib/python3.10/site-packages/gymnasium/envs/registration.py:520: UserWarning: WARN: Overriding environment HandManipulateBlockRotateZ-v0 already in registry.
  logger.warn(f"Overriding environment {new_spec.id} already in registry.")
/home/master-andreas/.local/lib/python3.10/site-packages/gymnasium/envs/registration.py:520: UserWarning: WARN: Overriding environment HandManipulateBlockRotateZ-v1 already in registry.
[... 239 lines cut ...]
```python
...
$ pip uninstall gymnasium_robotics
...
$ pip install gymnasium_robotics  
...
[master-andreas@master-pc ~]$ py                            
Python 3.10.8 (main, Nov  1 2022, 14:18:21) [GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import gymnasium_robotics
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/master-andreas/.local/lib/python3.10/site-packages/gymnasium_robotics/__init__.py", line 1, in <module>
    from gymnasium.envs.registration import register
  File "/home/master-andreas/.local/lib/python3.10/site-packages/gymnasium/__init__.py", line 12, in <module>
    from gymnasium.envs.registration import make, spec, register, registry, pprint_registry
  File "/home/master-andreas/.local/lib/python3.10/site-packages/gymnasium/envs/__init__.py", line 352, in <module>
    load_env_plugins()
  File "/home/master-andreas/.local/lib/python3.10/site-packages/gymnasium/envs/registration.py", line 309, in load_env_plugins
    fn = plugin.load()
  File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 173, in load
    return functools.reduce(getattr, attrs, module)
AttributeError: partially initialized module 'gymnasium_robotics' has no attribute 'register_robotics_envs' (most likely due to a circular import)
>>> 
```

rodrigodelazcano, 2022-12-25:
Two things:

## Installing from Github source
You shouldn't be working inside the repo `Gymnasium-Robotics` where your package directory is `gymnasium-robotics`. Try a python import from outside this directory, it should work that way

## Installing from Pypi
This will be fixed in the next release (after mamujoco and d4rl are merged). The issue is in the entrypoint in the `setup.py` file  https://github.com/Farama-Foundation/Gymnasium-Robotics/blob/bb73fe22abc855c18128f29ef8609fccf2e437b6/setup.py#L81. In the Github repo this is fixed but in the Pypi release version it's pointing to the function instead to the file:
```
entry_points={
        "gymnasium.envs": ["__root__ = gymnasium_robotics:register_robotics_envs"]
    },
```
This issues shouldn't affect the current use of Gymnasium-Robotics since all environments are meant to be used as registered environments in `Gymnasium`. However, is there something related to these issues that will be limiting the development of `mamujoco`?

Kallinteris-Andreas, 2022-12-27:
It, probably does not affect mamujoco, can you check the discord `robotics-dev`, for my module export issue, Thanks

closed 2022-12-27 by Kallinteris-Andreas