Farama-Foundation/Gymnasium-Robotics #95: [Bug Report] pip mamujoco install error

sebidoe, 2023-01-11:
**Describe the bug**
I tried installing the mamujoco environement and pip reports that the extra "mamujoco" is not provided by gymnasium-robotics

**Code example**
pip install gymnasium-robotics[mamujoco]

WARNING: gymnasium-robotics 1.2.0 does not provide the extra 'mamujoco'

**System Info**
Describe the characteristic of your environment:
 * gymnasium was installed with pip
 * OS Version: Ubuntu 22.04.1 LTS (Jammy Jellyfish)
 * Python 3.10

**Additional context**
Add any other context about the problem here.

### Checklist

- [x] I have checked that there is no similar [issue](https://github.com/Farama-Foundation/gymnasium-robotics/issues) in the repo

rodrigodelazcano, 2023-01-11:
Hi! We haven't included `mamujoco` environments in the current `1.2.0` release. We are waiting to perform some evaluation test before making a new release. Other than that, the documentation has to also be updated and remove the custom dependency installation for `mamujoco`. In the future the dependencies will be included with a normal pip installation `pip install gymnasium-robotics`. 

For now the environments have to be installed from source:

```
git clone https://github.com/Farama-Foundation/Gymnasium-Robotics.git
cd Gymnasium-Robotics
pip install -e.
```

I'll make a PR with a documentation update and leave this thread open until the next release is made.

Kallinteris-Andreas, 2023-05-17:
@rodrigodelazcano close

closed 2023-05-17 by rodrigodelazcano