Farama-Foundation/Gymnasium-Robotics #96: [Proposal] Add `terminate_when_unhealthy` option to mamujoco

sebidoe, 2023-01-11:
### Proposal 

Would it be possible to add the `terminate_when_unhealthy` option from the single agent version in Gymnasium to mamujoco?

### Motivation

For some experiments I would like to keep the agent running and even when the state becomes unhealthy.

### Pitch

Add an option to pass the `terminate_when_unhealthy` option to the underlying single agent environment through with the mamujoco parallel env creation.

### Checklist

- [x] I have checked that there is no similar [issue](https://github.com/Farama-Foundation/gymnasium-robotics/issues) in the repo

rodrigodelazcano, 2023-01-11:
This is a great suggestion. @Kallinteris-Andreas what would it look like to add the aditioanl `**kwargs` for each environment here https://github.com/Farama-Foundation/Gymnasium-Robotics/blob/81ec1d34cc1c3fddafbe4131a8ef23dd678527f0/gymnasium_robotics/envs/multiagent_mujoco/mujoco_multi.py#L110?

Will it break any functionality of the environments?

Kallinteris-Andreas, 2023-01-11:
No, it should not, break any functionality, would this require a version bump (or not because it is technically not released).

I should be able to work on it tomorrow, busy updating the DOCs today.

rodrigodelazcano, 2023-01-11:
We don't need to make a version bump since work for `mamujoco` is still in progress.

When you have time no worries.

Kallinteris-Andreas, 2023-01-23:
@sebidoe it should work now

Kallinteris-Andreas, 2023-03-07:
@sebidoe close this issue, unless you still face this issue

sebidoe, 2023-03-08:
Thanks for implementing this! 👍

closed 2023-03-08 by sebidoe