Farama-Foundation/Gymnasium-Robotics #113: [Question] `reset_mocapxbody()` in `mocap_set_action()`

matinmoezzi, 2023-02-16:
### Question

Hi, I have a robotic arm and am controlling it using a mocap body and `mocap_set_actoin()` provided in `mujoco_utils`. What I am observing is that every time I call `mocap_set_action`, `reset_mocapxbody()` is called inside it which leads to an undesirable movement of the end-effector as it is trying to change the mocap body position. I am wondering if `reset_mocapxbody` is necessary for `mocap_set_action`. I understand the purpose of this function but it causes an extra unwanted movement of the end-effector.
Thanks.

rodrigodelazcano, 2023-02-17:
It shouldn't be adding any extra movement to the end-effector body if the mocap is linked to it. Are you experiencing big displacements due to this reset function?

You are correct that this is a little redundant but it is a good sanity check to make sure that the mocap is at the same pose as the body it is linked to before adding the displacement action to the mocap.

matinmoezzi, 2023-02-17:
Hi @rodrigodelazcano,
Thank you for your response.
Yes. I understand what the reset function does. But, any changes in the position of the mocap result in a movement of the body because of the weld constraint. In fact, this procedure is the basis of `mocap_set_action`: changing mocap_pos -> changing end-effector pos
Moreover, I have experienced that the mocap pos could not be the same as the body pos since every mocap displacement could not lead to the body displacement because of inverse kinematics. Please correct me if my understanding is wrong.

rodrigodelazcano, 2023-02-19:
@matinmoezzi yes you are correct, Mujoco will prioritize having a feasible IK configuration before following the mocap body (this is the reason the mocap is reset every step, to make sure that it doesn't grow to unachievable configurations). 
A better way to implement the actions will be to calculate the new end-effector pose with the current end-effector body pose and then update the mocap's pose. However, I still don't see why resetting the mocap pose gives extra displacements.

Do you have the time to implement this changes in a PR?