Farama-Foundation/Gymnasium-Robotics #122: [Question] Adapting Fetch Robot for the FrankaKitchen Task

aalmuzairee, 2023-02-28:
### Adapting Fetch Robot for the FrankaKitchen Task

Hi, 
I'm trying to add the Fetch Robot from the Fetch Environment into the Franka-Kitchen Task. I noticed that the dynamics (damping, friction) of the Fetch Robot are different from the ones in the urdf supplied by: https://github.com/ZebraDevs/fetch_ros/blob/melodic-devel/fetch_description/robots/fetch.urdf and also different from the Franka Robot. How can I make it as realistic as possible and also make it functional with the inverse kinematics library (similar to the Franka)?

Appreciate any suggestions.

rodrigodelazcano, 2023-03-01:
Hey @aalmuzairee ,
Regarding the model selection for each robot:
1. The Fetch robot mjcf model is the original from https://openai.com/research/ingredients-for-robotics-research and I don't know how they came up with the dynamics and how realistic they intended it to be (I think this simulation is more focused for algorithm testing purposes than sim2real). However, it would be very interesting to make a new environment version with a more acurate robot model. Currently we are getting our new robot models from the official mujoco repository [mujoco_menagerie](https://github.com/deepmind/mujoco_menagerie). I think the first step should be converting the `urdf` model to mjcf mujoco compatible and make a PR to the mujoco_menagerie. Then we can see how to incorporate it to `gymnasium-robotics`
2. The Franka mjcf model has been adapted from `mujoco_menagerie`. Can you post an issue on there repo specifying the difference?

Then to create your custom environment I'm afraid that you'll have to create your own mjcf model and environment class. You can base your environment in the current Franka implementation but there is no utility functions to replace robots in an environment.

Having said this I'm happy to give you a hand creating the Fetch mjcf model from the official fetch repo.

aalmuzairee, 2023-03-07:
Hi @rodrigodelazcano,

Appreciate your input! This was enlightening. I'm currently working on converting an accurate Fetch model to an mjcf format. If I encounter any roadblocks, I'll be sure to let you know. Thanks for the help.

closed 2023-03-07 by aalmuzairee