Farama-Foundation/Gymnasium-Robotics #126: [Bug Report] FetchReach-v2 render bug

frietz58, 2023-03-14:
The `FetchReach-v2` crashes when initialized with `render_mode="human" (this does not happen for the other three fetch environments):

```python
import gymnasium as gym

env = gym.make("FetchReach-v2", render_mode="human")
env.reset()
terminated = False

while not terminated:
    obs, reward, terminated, truncated, info = env.step(env.action_space.sample())
```

results in 
```bash
Traceback (most recent call last):
  File "/home/finn/PycharmProjects/icml-2023-code/algos/sql_v2/robotics-gym.py", line 4, in <module>
    env.reset()
  File "/home/finn/PycharmProjects/icml-2023-code/venv/lib/python3.9/site-packages/gymnasium/wrappers/time_limit.py", line 69, in reset
    return self.env.reset(**kwargs)
  File "/home/finn/PycharmProjects/icml-2023-code/venv/lib/python3.9/site-packages/gymnasium/wrappers/order_enforcing.py", line 43, in reset
    return self.env.reset(**kwargs)
  File "/home/finn/PycharmProjects/icml-2023-code/venv/lib/python3.9/site-packages/gymnasium/wrappers/env_checker.py", line 45, in reset
    return env_reset_passive_checker(self.env, **kwargs)
  File "/home/finn/PycharmProjects/icml-2023-code/venv/lib/python3.9/site-packages/gymnasium/utils/passive_env_checker.py", line 208, in env_reset_passive_checker
    result = env.reset(**kwargs)
  File "/home/finn/PycharmProjects/icml-2023-code/venv/lib/python3.9/site-packages/gymnasium_robotics/envs/robot_env.py", line 184, in reset
    self.render()
  File "/home/finn/PycharmProjects/icml-2023-code/venv/lib/python3.9/site-packages/gymnasium_robotics/envs/robot_env.py", line 404, in render
    self._get_viewer(self.render_mode).render()
  File "/home/finn/PycharmProjects/icml-2023-code/venv/lib/python3.9/site-packages/gymnasium_robotics/envs/robot_env.py", line 427, in _get_viewer
    self._viewer_setup()
  File "/home/finn/PycharmProjects/icml-2023-code/venv/lib/python3.9/site-packages/gymnasium_robotics/envs/fetch/fetch_env.py", line 241, in _viewer_setup
    lookat = self.get_gripper_xpos()
AttributeError: 'MujocoPyFetchReachEnv' object has no attribute 'get_gripper_xpos'
```
Simply replacing `lookat = self.get_gripper_xpos()` with `lookat = self._get_gripper_xpos()` results in the environment executing as expected without error.

rodrigodelazcano, 2023-03-14:
Hi @frietz58, this bug should be fixed in the next release of `Gymnasium-Robotics`. Is it possible for you to work with a source installation in the mean time?

frietz58, 2023-03-16:
Sure, all is fine, I just wanted to report this :)

closed 2023-03-16 by frietz58