Farama-Foundation/Gymnasium-Robotics #135: [Question] How can I fix the orientation (just like FetchEnv) of Franka Panda in Franka kitchen?

RayYoh, 2023-03-25:
### Question

Hi, thanks for your nice work.
I would like to ask how can I fix the orientation (just like FetchEnv) of Franka Panda in Franka kitchen.
I have try to set the `action[3:7]=[0, 0, 0]` of Franka Panda, but when I render the environment I see the orientation still changed. I also print the achieved pose of the EEF site in `step()` after:
```
for _ in range(self.control_steps):
                delta_qpos = self.controller.compute_qpos_delta(
                    target_eef_pose, target_orientation
                )
                ctrl_action[:7] = self.data.ctrl.copy()[:7] + delta_qpos[:7]

                # Do not use `do_simulation`` method from MujocoEnv: value error due to discrepancy between
                # the action space and the simulation control input when using IK controller.
                # TODO: eliminate error check in MujocoEnv (action space can be different from simulaton control input).
                self.data.ctrl[:] = ctrl_action
                mujoco.mj_step(self.model, self.data, nstep=self.frame_skip)

                if self.render_mode == "human":
                    self.render()
```
I see the orientation indeed changed.
I know mocap may help me in this task (need to modify the xml file), but I really want to know why the orientation changed. Is it because of the IK controller?
By the way, what is the meaning of `self.control_steps`.

Kallinteris-Andreas, 2023-12-30:
@rodrigodelazcano was the issue fixed with https://github.com/Farama-Foundation/Gymnasium-Robotics/pull/145

rodrigodelazcano, 2023-12-31:
@Kallinteris-Andreas yes , the IK control was removed in that PR.

closed 2023-12-31 by rodrigodelazcano

shaoyifei96, 2025-01-20:
Hello! Is there another way to do inverse kinematics internally or do we need to find our own solution