Farama-Foundation/Gymnasium-Robotics #155: [Bug Report] Maze/AntMaze issues

alexdavey, 2023-06-15:
Hi, I believe I have found a couple of issues in the Maze/AntMaze environments. I have resolved both of these issues in [commit 5573d5e](https://github.com/Farama-Foundation/Gymnasium-Robotics/compare/main...alexdavey:Gymnasium-Robotics:antmaze), and I’m happy to submit a PR.

### 1) AntMaze sparse reward always zero

For continuing tasks in AntMazeEnv, the sparse reward is always zero. This is because at each step, `.compute_terminated()` resets the goal when the Ant is sufficiently close, before the reward is calculated.

Here’s a code example to test this:

```python
import numpy as np
import gymnasium as gym

env = gym.make("AntMaze_UMaze-v3", continuing_task=True, reward_type="sparse")

total_reward = 0

# 1000 "episodes" of 100 steps
for _ in range(1000):
    env.reset()

    for _ in range(100):
        action = env.action_space.sample()
        _, rew, _, _, _ = env.step(action)
        total_reward += rew

print(total_reward)

```

Currently this returns exactly zero since no reward is collected, but placing `.compute_reward()` above `.compute_terminated()` gives a non-zero reward. In PointMaze, this issue was fixed in commit ace181e2a0d83be8940103bbb4d7e917afb30119.

### 2) AntMaze can reset into a terminal state

The Ant will sometimes start within the goal radius. This is because there is a `maze_size_scaling` factor missing in the distance check in `MazeEnv.generate_reset_pos()`.

In AntMaze `maze_size_scaling = 4`, so the xy position noise can be up to 1.0 in each direction. An unfortunate combination of goal noise and reset noise can cause the Ant to start within 0.45 of `self.goal`. This issue does not affect PointMaze because there `maze_size_scaling = 1`.

Here’s a code example to test this:
```python
import numpy as np
import gymnasium as gym

env = gym.make("AntMaze_UMaze-v3", continuing_task=True)

for _ in range(1000):
    obs, _ = env.reset()
    dist = np.linalg.norm(obs["achieved_goal"] - obs["desired_goal"])
    assert dist > 0.45
```

### Checklist

- [x] I have checked that there is no similar [issue](https://github.com/Farama-Foundation/gymnasium-robotics/issues) in the repo (**required**)

Kallinteris-Andreas, 2023-06-16:
1.
Your test for sparse rewards, is not correct, since using random actions is very unlikely to get you to the terminal state

2.
Does this also affect the `PointEnv`?

alexdavey, 2023-06-16:
Thanks for taking a look at this issue.

1. I agree that it is not a robust test in general, it’s primarily just a check to compare pre/post commit (after the suggested changes, it does get a very small but non-zero reward). If you have suggestions for a more robust test I can implement that, but I think the issue is fairly clear from the step method of AntMaze:

https://github.com/Farama-Foundation/Gymnasium-Robotics/blob/d0a10dc8b5b919b21572b5efc5b45ba98f452553/gymnasium_robotics/envs/maze/ant_maze.py#L277-L289

and comparing to the PointMaze step:

https://github.com/Farama-Foundation/Gymnasium-Robotics/blob/d0a10dc8b5b919b21572b5efc5b45ba98f452553/gymnasium_robotics/envs/maze/point_maze.py#L379-L391

`.compute_terminated()` is common to both and resets the goal location if within 0.45.

This issue previously existed for PointMaze. I’m proposing to make the same fix as commit ace181e, but for AntMaze.

2. The PointEnv/PointMaze env’s are not affected by either of these issues, due to the commit referenced above, and because `maze_size_scaling = 1` so the missing factor does not change anything.

Kallinteris-Andreas, 2023-06-16:
1. I would say it is worrying that `compute_terminated` is not a pure function, it should be refactored into 2 function:`
compute_terminated` (but now it just computes if it is terminated)

These changes would requite a new revision "AntMaze_UMaze-v4"

@rodrigodelazcano

Kallinteris-Andreas, 2023-06-16:
@alexdavey feel free to make a PR

PR #156 (2023-06-22): Maze/AntMaze issues
# Description

Fixes #155 

* refactored `compute_terminated` so it is now a pure function, added `update_goal`

* added info["success"] key to AntMaze

* fixed sparse reward issue in AntMaze

* fixed reset issue in Maze

* added test for reset issue

* bumped version to AntMaze-v4

## Type of change

- [x] Bug fix (non-breaking change which fixes an issue)
- [x] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work as expected)
- [x] This change requires a documentation update

# [... cut]
closed 2023-06-22 by rodrigodelazcano