Farama-Foundation/Gymnasium-Robotics #166: [Bug Report] The `tasks_to_complete` variable does not reset properly in `FrankaKitchen-v1`

greg3566, 2023-08-16:
**Describe the bug**
The `tasks_to_complete` variable does not reset properly in `FrankaKitchen-v1`. The `info['tasks_to_complete']` returned from the `reset()` function does match the `tasks_to_complete` argument passed when initializing the environment. However, in subsequent calls to the `step()` function, the `info['tasks_to_complete']` excludes tasks removed in the previous episode. Also, no reward will be generated from the removed tasks.

**Code example**
```
env = gym.make('FrankaKitchen-v1', tasks_to_complete=['microwave', 'kettle'])
```
After creating the environment, complete the kettle task in an episode and then execute the following code:
```
obs, info = env.reset()
print(info['tasks_to_complete'])
obs, reward, terminated, truncated, info = env.step(env.action_space.sample())
print(info['tasks_to_complete'], info['episode_task_completions'])
```
This will output:
```
{'kettle', 'microwave'}
['microwave'] []
```

**Solution**
Change `task_to_complete` to `tasks_to_complete` in `reset` function of `kitchen_env`:
```
        self.tasks_to_complete = set(self.goal.keys())
        info = {
            "tasks_to_complete": self.tasks_to_complete,
            "episode_task_completions": [],
            "step_task_completions": [],
        }
```

**System Info**
- pip install within a conda env
- Ubuntu 20.04
- Python 3.8

### Checklist

- [x] I have checked that there is no similar [issue](https://github.com/Farama-Foundation/gymnasium-robotics/issues) in the repo (**required**)

rodrigodelazcano, 2023-09-02:
Thank you for finding this bug @greg3566. I can replicate the issue and your solution solves it. Do you have the time to make a PR with the fix?

Also, thanks to your issue I've also realized that in the same step that a task is completed, that task still appears in the `task_to_complete` info item of that specific step. The task is removed in the info's of the following steps but it shouldn't appear in the info returned of the step when it was achieved. I'll fix this in another PR.

rodrigodelazcano, 2023-09-04:
Fixed in https://github.com/Farama-Foundation/Gymnasium-Robotics/pull/169

closed 2023-09-04 by rodrigodelazcano