Farama-Foundation/Gymnasium-Robotics #173: [Question] How to get the absolute position and velocities in the global coordinate system

dchen48, 2023-09-08:
I am working on multi-agent reacher with 2 links. How can I get the absolute position and velocities in the global coordinate system? Thank you for the help.

Kallinteris-Andreas, 2023-09-09:
```
env.single_agent_env.unwrapped.data.qpos
env.single_agent_env.unwrapped.data.qvel
```
You can check the doc pages of single agent gymnasium environments, https://gymnasium.farama.org/main/environments/mujoco/reacher/

dchen48, 2023-09-09:
Thanks a lot for the quick reply. Can you please help me verify:

1. env.single_agent_env.unwrapped.data.body("fingertip").subtree_linvel return the absolute velocity of the fingertip in the global coordinate system.
2. env.single_agent_env.unwrapped.data.body("fingertip")..xpos return the absolute position of the fingertip in the global coordinate system.

Your help is greatly appreciated.

Kallinteris-Andreas, 2023-09-09:
1. what do you mean by "velocity in the global coordinate system"  do you mean "Cartesian velocity" if so you will need to use the second part of `cvel` (more on https://mujoco.readthedocs.io/en/latest/APIreference/APItypes.html?highlight=cvel#mjdata)
2. yup looks good

dchen48, 2023-09-09:
Thanks for the help. Yes, I mean "Cartesian velocity", and now I understand how to get it.

closed 2023-09-09 by dchen48