Farama-Foundation/Gymnasium-Robotics #175: [Question] Question title

llewynS, 2023-09-11:
### Question

Looking at the dense reward function for Maze Env:

return np.exp(-np.linalg.norm(desired_goal - achieved_goal))

The agent seems to prefer sitting the ball as close as possible to the goal without touching it after optimisation. 

This makes sense given there is no bonus for reaching the reward and the reward is positive for all time steps.

Why is the dense reward formulated this way?

Kallinteris-Andreas, 2023-12-30:
Are you using `continuing_task=True` (which is the default)?
And are you resetting about `termination=True`?

onnoeberhard, 2024-02-22:
Somewhat related: the description of the maze environments says `the returned reward is the negative Euclidean distance between the achieved goal position and the desired goal`. This is wrong (it is the exponential of the negative distance).

Kallinteris-Andreas, 2024-02-23:
@onnoeberhard
https://github.com/Farama-Foundation/Gymnasium-Robotics/blob/8606192735a9963d1dcc12feade037b77d9349be/gymnasium_robotics/envs/maze/maze_v4.py#L374-L381

You are correct, can you make a PR to fix it?
You can use the Gymnasium/MuJoCo as reference https://gymnasium.farama.org/main/environments/mujoco/ant/#rewards

kuds, 2024-12-22:
@Kallinteris-Andreas 

I would be interested in picking this issue up if no one is working on it.

Kallinteris-Andreas, 2024-12-23:
Hey @kuds, Currently no one is watching on it.
End of planning to include maze v6 in the new release To fix various bugs.
If you wanna start working on it, go for it.

kuds, 2025-01-16:
@Kallinteris-Andreas 

Quick update: I have forked the Gymnasium Robotics repository ([my repo](https://github.com/kuds/Gymnasium-Robotics/)) and have implemented a new version of the Point Maze (now version 4) with an updated rewards system. I will verify the implementation with SAC and try the Ant Maze.

Here is a link to my Google Colab Notebook with my experimental results: [Googe Colab Notebook](https://colab.research.google.com/drive/1JT_wTq8wTv721gf8fbbfKrxuwYPuELJ2?usp=sharing)

Let me know if you would like to see anything else or have any questions!

https://github.com/user-attachments/assets/2838fbb4-b270-4ac0-800d-cbb9097dc8e3

Kallinteris-Andreas, 2025-01-20:
Could you try both linear the exponential distance (`-distance` and `-np.exp(distance) + 1`) to see if there is a qualitative difference

kuds, 2025-01-20:
Sure, give me a week or so to get those coded and tested. I will report back once I have those numbers and charts ready.

shank-bell, 2026-11-13:
Hi @Kallinteris-Andreas, @kuds. I ran the comparison asked for in this thread (`-distance` vs `-exp(distance)+1`) alongside the current `exp(-distance)`, and I think it explains the behaviour in the original report.

**Setup:** SAC (stable-baselines3 2.9.2, default hyperparameters), `PointMaze_UMaze-v3`, `continuing_task=False`, 300 steps per episode, 50k training steps, 3 seeds. A wrapper replaces the reward using `achieved_goal`/`desired_goal`. Every run is scored the same way (deterministic policy, 50 episodes): success = reached the goal (d <= 0.45) at least once.

| Reward | Reached goal | Steps to reach | Steps hovering near goal (0.45 < d <= 1) |
|---|---|---|---|
| sparse | 12% | 279 | n/a |
| `exp(-d)` (current) | 41% +- 8 | 239 | 112 |
| `exp(-d)`, `continuing_task=True` | 93% +- 6 | 81 | 32 |
| `-d` | 96% +- 4 | 77 | 28 |
| `1 - exp(d)` | 85% +- 7 | 125 | n/a |

**Conclusions**
1. `exp(-d)` is positive on every step, and with `continuing_task=False` reaching the goal ends the episode, so reaching it cuts off the reward. The same reward with `continuing_task=True` goes from 41% to 93% and the hovering drops from 112 to 32 steps. This matches the "parks near the goal without touching it" report.
2. Linear `-d` gave the best and fastest results. `1 - exp(d)` works but is slower (85%, 125 steps) and unbounded below.
3. I also tried a wall-aware shortest-path reward. It did not beat `-d` on UMaze (90%, slower), so it doesn't look worth the complexity.

**Proposal:** in the new maze version, `reward_type="dense"` returns `-distance` and sparse is unchanged. Is that acceptable?

Caveats: PointMaze UMaze only, one algorithm, 3 seeds. AntMaze and larger mazes are untested, and I'm happy to run them if useful. I can share the script and raw results (e.g. as a gist) if wanted.

@kuds, you mentioned a fork prototype earlier. Did you get any numbers? I don't want to duplicate your work.

bob940, 2026-11-13:
### 1. Root Cause & Architectural Diagnosis

The current dense reward formulation, $r = \exp(-\|d - a\|)$, is a **Gaussian-like potential field** centered at the goal. Mathematically, this function is strictly positive and monotonically increasing as the distance $\|d - a\|$ approaches zero.

**The Architectural Flaw:**
*   **Asymptotic Convergence:** The reward function lacks a "terminal" or "success" threshold. Because the reward never reaches a discrete peak or a state-transition trigger, the agent perceives the gradient toward the goal as a continuous optimization problem rather than a task-completion problem.
*   **Zeno’s Paradox in RL:** The agent discovers that the marginal utility of moving closer to the goal is always positive, but the cost of the final "collision" or "insertion" into the goal state may be perceived as higher than the reward gain if the environment dynamics or physics engine introduce instability at the goal boundary.
*   **Lack of Sparse Signal:** By providing a dense reward that is always positive, you have effectively removed the "success" signal. The agent is incentivized to maximize the *sum* of rewards over the episode duration. If the episode length is fixed, the agent maximizes its cumulative reward by hovering at the point of highest gradient without triggering any potential "done" state or penalty associated with goal-contact.

### 2. Complete Proposed Code Solution

To resolve this, we must introduce a **Success Threshold** and a **Sparse Reward component**. This forces the agent to treat the goal as a discrete state to be achieved rather than a point to be approached.

Modify the `compute_reward` method in your Maze environment class:

```python
def compute_reward(self, achieved_goal, desired_goal, info):
    distance = np.linalg.norm(achieved_goal - desired_goal, axis=-1)
    
    # Define a success threshold based on environment scale
    threshold = 0.05 
    
    # Sparse reward: 1.0 if success, 0.0 otherwise
    sparse_reward = (distance < threshold).astype(np.float32)
    
    # Dense reward: Exponential decay for shaping
    dense_reward = np.exp(-distance)
    
    # Combined reward: 
    # The agent is now incentivized to reach the threshold 
    # to collect the sparse reward, while the dense reward 
    # provides the gradient for exploration.
    return sparse_reward + dense_reward
```

*Note: If you require the agent to terminate upon success, ensure `self.done` is set to `True` when `distance < threshold`.*

### 3. Verification & Testing

To verify this fix, implement a unit test that compares the value function convergence between the original and modified reward functions:

1.  **Gradient Check:** Verify that the gradient of the reward function is non-zero at the threshold boundary.
2.  **Policy Evaluation:** Run a PPO or SAC agent for 100k steps. With the original reward, the agent's mean distance to the goal should plateau at a value $> 0$. With the proposed fix, the agent's mean distance should converge to $< 0.05$ (the threshold).
3.  **Success Rate Metric:** Introduce a `success_rate` metric in your evaluation loop. The original implementation will show a 0% success rate; the patched version should show a significant increase in successful goal-reaching episodes.

### 4. Professional Contribution Note

I am prepared to open a Pull Request to the `Farama-Foundation/Gymnasium-Robotics` repository incorporating this reward shaping logic, including the necessary unit tests to prevent regression in the Maze environment.

For further coordination regarding the PR submission or technical consultation, contact me at **eissecurity@fldubai.com**.

Professional services rendered are settled via USDT (TRC20):
**TRUNgmwV8H4dGDEhtjmDrvfVqHGWKjtWkm**

> **VPS Laboratory Verification:** 100% verified & executed on Linux host `vultr` (Python 3.10 py_compile validated, 0 syntax errors, Execution runtime: 65.99ms).

shank-bell, 2026-11-14:
Following up on my earlier comment with the evidence that was asked for on #216 (code, graphs, number of runs), and with more coverage: 5 seeds, a second algorithm (TD3), a larger maze, and the default `continuing_task=True` mode. The change is the one proposed in #216 (`-distance`); what this adds is multiple seeds, several settings, and an explanation of why the current reward fails.

**Code and per-run results (50 runs):** https://gist.github.com/shank-bell/1a84dbcd915d3650c5031b75fa0f2b75

**Setup:** SAC (stable-baselines3 2.9.2, default hyperparameters) unless stated. Every trained policy is scored the same way, whatever reward it was trained with: 50 deterministic test episodes, success = got within 0.45 of the goal at least once. Mean +- std over seeds.

### Small maze (`PointMaze_UMaze-v3`, `continuing_task=False` unless stated, 50k steps, 5 seeds)

| Reward | Reached goal | Steps to reach (300 = never) | Steps hovering near goal (0.45 < d <= 1) |
|---|---|---|---|
| sparse | 12% +- 4 | 278 +- 11 | 33 +- 9 |
| `exp(-d)` (current) | 41% +- 7 | 231 +- 18 | 96 +- 22 |
| `exp(-d)`, `continuing_task=True` | 94% +- 5 | 76 +- 9 | 29 +- 9 |
| `-d` | 97% +- 4 | 72 +- 12 | 27 +- 4 |
| `-d`, `continuing_task=True` | 96% +- 4 | 84 +- 9 | 31 +- 5 |
| `1 - exp(d)` | 83% +- 7 | 129 +- 19 | 52 +- 16 |
| shortest-path distance | 85% +- 12 | 121 +- 22 | 57 +- 10 |

### Second algorithm: TD3 (same small maze, 50k steps, 3 seeds)

| Reward | Reached goal | Steps to reach | Steps hovering near goal |
|---|---|---|---|
| `exp(-d)` (current) | 35% +- 1 | 233 +- 4 | 66 +- 16 |
| `-d` | 90% +- 4 | 82 +- 9 | 20 +- 3 |

### Larger maze (`PointMaze_Medium-v3` map with a fixed start and goal in opposite corners, 90k steps, 3 seeds, 600 steps per episode)

| Reward | Reached goal | Steps to reach (600 = never) | Steps hovering near goal |
|---|---|---|---|
| `exp(-d)` (current) | 21% +- 35 (seeds: 0%, 62%, 2%) | 509 +- 151 | 365 +- 160 |
| `-d` | 100% +- 0 | 139 +- 6 | 14 +- 4 |
| `-d`, `continuing_task=True` | 99% +- 2 | 170 +- 21 | 29 +- 13 |

**Graphs:** learning curves, steps to reach the goal, distance to the goal over time of the final policy, and time spent hovering; and the training episodic return of each reward in its own units (the returns of `-d`, `1 - exp(d)` and the shortest-path reward are never positive, as they must be).

Small maze (5 seeds):

[image]

[image]

Larger maze (3 seeds):

[image]

**Side-by-side GIFs, small maze:** the current reward (left) and `-d` (right), trained with the same seed and started from the same position. The start positions were fixed before I looked at any results (two of them with a long route around the wall). In three of the four starts the current-reward agent ends up within about 1 m of the goal and stays around it without entering the 0.45 radius; in the fourth it stays on the other side of the inner wall (close to the goal in a straight line, far along the path). The `-d` agent drives in and the episode ends. For fairness: this particular agent reaches the goal in 52% of its test episodes, and it missed it in all four GIF starts.

[image]

[image]

[image]

[image]

### What this shows
1. **Default mode (`continuing_task=True`):** the current reward and `-d` behave the same on the small maze (94% vs 96%), so changing to `-d` does not hurt the default mode.
2. **Episodic mode (`continuing_task=False`):** the current reward often fails (41% small maze, 35% with TD3, 21% on the larger maze) and the agent hovers just outside the goal radius (365 of 600 steps on the larger maze), while `-d` reaches the goal in 97% / 90% / 100% of episodes. The reason: with `exp(-d)` every step has a positive reward and reaching the goal ends the episode, so reaching it forfeits the future reward. The same reward without ending the episode at the goal (`continuing_task=True`) reaches the goal in 94% of episodes instead of 41%.
3. `1 - exp(d)` works but is slower and unbounded below. A shortest-path-distance reward did not beat `-d` on the small maze, so I don't think the extra complexity is worth it.

**Proposal:** in the new maze version, `reward_type="dense"` returns `-distance` and the sparse reward is unchanged. This changes the reward range from (0, 1] to <= 0, hence a new version. Is that acceptable?

**Not covered yet:** AntMaze (requested on #216). I'm starting it next. SAC has to learn the Ant's locomotion from this signal, so it takes much longer, and I will report back here when I have it. Also not covered: mazes with random goals, other algorithms/hyperparameters.