Farama-Foundation/Gymnasium-Robotics #181: [Bug Report] AntEnv obs[:2] is not x, y but z coordinate and x-orientation

llewynS, 2023-09-30:
**Describe the bug**
The _get_obs function does the following transformation to the observation space
        achieved_goal = ant_obs[:2]
        observation = ant_obs[2:]
But reading AntEnv V4 the pos 0 and pos 1 observations are not the final x and y coordinates. As the antenv is directly queried and no computation is done in the environment it's clear that this is an error.

### Checklist

- [ ] I have checked that there is no similar [issue](https://github.com/Farama-Foundation/gymnasium-robotics/issues) in the repo (**required**)

Kallinteris-Andreas, 2023-09-30:
the first 2 are the X&Y coordinates of the `torso`
https://gymnasium.farama.org/main/environments/mujoco/ant/#observation-space

llewynS, 2023-09-30:
From your link this is not correct. First two are the z coordinate and the x-orientation as I stated.
[image]

Kallinteris-Andreas, 2023-09-30:
Read the description above the table (and the excluded observations are in the bottom of the table)

closed 2023-09-30 by llewynS