Farama-Foundation/Gymnasium-Robotics #182: [Question] Is there a way to set the environment to a specific state in franka kitchen tasks?

ygtkorkmaz, 2023-10-04:
I am trying to replay the expert demonstrations in the franka kitchen environment but I couldn't find any way of setting the environment to a specific state in the code.  Any help would be appreciated!

Kallinteris-Andreas, 2023-10-05:
Set the values in 
```py
env.unwrapped.data.qpos
env.unwrapped.data.qvel
```

more info how mujuco state works can be found here: https://mujoco.readthedocs.io/en/latest/computation/index.html#the-state

ygtkorkmaz, 2023-10-05:
Got it, thanks a lot. Also, I realized envs in gymnasium-robotics don't have a get_dataset() function. I believe I should get the dataset using Minari?

Kallinteris-Andreas, 2023-10-05:
yes, this project only have robotic environments, for offline RL you need to use minari

ygtkorkmaz, 2023-10-05:
Great, thank you so much!

closed 2023-10-05 by ygtkorkmaz