Farama-Foundation/Gymnasium-Robotics #184: [Bug Report] Temporary XML file is written to fixed path

nicoguertler, 2023-10-06:
Hi, 

I would like to report a behavior of the Maze class that I believe to be a bug.

**Describe the bug**
The temporary XML file is always written to the same fixed path (on my system `/tmp/ant_maze.xml`). That leads to crashes if several processes try to instantiate an environment. 

**Code example**
```
import gymnasium as gym

env = gym.vector.AsyncVectorEnv([lambda: gym.make("PointMaze_Open-v3") for _ in range(10)])
```
This produces
```
ValueError: XML parse error 13:
Error=XML_ERROR_EMPTY_DOCUMENT ErrorID=13 (0xd) Line number=0
```
among other parsing errors.

**System Info**
 *  I am running Ubuntu 20.04.06
 * Gymnasium was installed via pip.
 * Python 3.8.10

**Additional context**
It looks like the problem is caused by the use of `path.dirname(tmp_dir)` [here](https://github.com/Farama-Foundation/Gymnasium-Robotics/blob/3a579d6efa2056cce394ffeb5d1ef83a58fea23a/gymnasium_robotics/envs/maze/maze.py#L154) which effectively discards the temporary directory created in the line above.

Thank you for your help!

### Checklist

- [x] I have checked that there is no similar [issue](https://github.com/Farama-Foundation/gymnasium-robotics/issues) in the repo (**required**)

nicoguertler, 2023-10-06:
On second thought I can see that this behavior is probably intended because the temporary directory gets removed again, sorry. I think it is still problematic though and would like to inquire if there is a chance that this problem will be solved.

roger-creus, 2023-10-10:
I'm facing the same issue when running the code in a cluster. The funny thing is that it won't always crash, only sometimes. Please let me know if you fix this issue.

roger-creus, 2023-10-11:
This solves it @Kallinteris-Andreas ! Thank you!

nicoguertler, 2023-10-11:
Works perfectly for me. Thank you very much @Kallinteris-Andreas !

Kallinteris-Andreas, 2023-10-19:
@rodrigodelazcano close

rodrigodelazcano, 2023-10-19:
Hey @Kallinteris-Andreas I just wanted to make sure that this issue is solved in every affected environment (mamujoco) as you mentioned in https://github.com/Farama-Foundation/Gymnasium-Robotics/pull/185#issuecomment-1768987758

closed 2023-12-30 by Kallinteris-Andreas