Farama-Foundation/Gymnasium-Robotics #203: [Question] In Maze environment, how to add multiple goals that exist simultaneously

Dhanushvarma, 2024-01-05:
### Question
In Maze environment, how to add multiple goals that exist simultaneously. Currently, the env is designed in such a manner that the goal spawns in the location that are set to "G", how can I make it such that there exists multiple goals that exist simultaneously?

Kallinteris-Andreas, 2024-01-06:
There can be multiple goals in the same maze, and one of them will be randomly selected when the environment is reset.

The simplest possible implementation would be to have a sequence of goals, that after one is complete the next one is activated

Having multiple goals at the same time might not work with the GoalEnv API

Dhanushvarma, 2024-01-07:
From what I see, the GoalEnv does not have anything to with single goal at a timepoint, do you think, if I modify the MazeEnv Class, it can be implemented?

Kallinteris-Andreas, 2024-01-07:
From what I have checked, there are 0 restrictions on what `desiredGoal` should be, so yes you can implement that in your fork without breaking the API

closed 2024-01-08 by Kallinteris-Andreas