Farama-Foundation/Gymnasium-Robotics #214: [Question] How to set camera to goal position?

AlexandreBrown, 2024-03-14:
### Question
Hi, I'm working on a goal-conditioned RL approach and would like to know if it is possible to move the camera of the environment to or near the generated goal of an environment (eg: `AntMaze_UMaze-v4`).  
  
My objective is to move the camera to the goal to capture the observation (pixels) near the goal.  
  
Thanks

Kallinteris-Andreas, 2024-03-14:
You have asked two separate questions.

1) move camera 
You will have to create a new camera And then use the `camera_name` argument.

https://gymnasium.farama.org/main/environments/mujoco/#rendering-arguments
( For creating a new camera you have to look to the MuJoCo documentation )

2) move the "Agent"
'I assume by that you mean to move the robot, To do so you simply have to change the position state (`env.unwrapped.data.qpos`)

Hope this helps.

AlexandreBrown, 2024-03-15:
@Kallinteris-Andreas Thanks a lot for the quick feedback.  
Your suggestion for moving the agent worked flawlessly.  
Thanks again!

closed 2024-03-15 by AlexandreBrown