Farama-Foundation/Gymnasium-Robotics #226: [Bug Report] MazeEnv Goal Used is in i,j but should be x,y

llewynS, 2024-09-03:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

**Describe the bug**
The goal in generate_target_goal is selected from maze.unique_goal_locations which is in i,j coordinates and is never converted to x,y so the goal will be in the wrong coordinate system for comparison. 

**Code example**
It's in the source code:

```
    def generate_target_goal(self) -> np.ndarray:
        assert len(self.maze.unique_goal_locations) > 0
        goal_index = self.np_random.integers(
            low=0, high=len(self.maze.unique_goal_locations)
        )
        goal = self.maze.unique_goal_locations[goal_index].copy()
        return goal
```
called in reset as 
```
if options is None:
            goal = self.generate_target_goal()
            # Add noise to goal position
            self.goal = self.add_xy_position_noise(goal)
```
and add_xy_position_noise(goal) does not convert to x,y

**System Info**
This is reading the source code in this repository.

### Checklist

- [X ] I have checked that there is no similar [issue](https://github.com/Farama-Foundation/gymnasium-robotics/issues) in the repo (**required**)

llewynS, 2024-09-03:
Actually further reading it just looks like the comments indicating that unique_goal_location are in (i,j) are incorrect and the comment should simply read (x,y) instead.

Kallinteris-Andreas, 2024-09-10:
so this is a comment only issue?