Farama-Foundation/Gymnasium-Robotics #227: [Question] `agent_obsk` in mujoco_multi.py

xihuai18, 2024-09-28:
### Question

https://github.com/Farama-Foundation/Gymnasium-Robotics/blob/b1acee9aea84f0952726918f7290ffa2a95eceaa/gymnasium_robotics/envs/multiagent_mujoco/mujoco_multi.py#L93-L98
                
`agent_obsk` in the original repo controls only the observation construction, but not the action factorization. I am wondering why implementing agent 0 performs all the actions when `agent_obsk` is None here.

Kallinteris-Andreas, 2024-09-28:
The behavior is the same as the original repo when `agent_obsk=None`, it simply becomes a single agent environment, it was just miss documented 

https://github.com/schroederdewitt/multiagent_mujoco/blob/b212ddd74b258e7cea006ff1d642b5ffada4b99d/multiagent_mujoco/mujoco_multi.py#L43

https://github.com/schroederdewitt/multiagent_mujoco/blob/b212ddd74b258e7cea006ff1d642b5ffada4b99d/multiagent_mujoco/mujoco_multi.py#L67

xihuai18, 2024-09-28:
I think the main difference is 

https://github.com/schroederdewitt/multiagent_mujoco/blob/b212ddd74b258e7cea006ff1d642b5ffada4b99d/multiagent_mujoco/mujoco_multi.py#L108-L116

and

https://github.com/Farama-Foundation/Gymnasium-Robotics/blob/b1acee9aea84f0952726918f7290ffa2a95eceaa/gymnasium_robotics/envs/multiagent_mujoco/mujoco_multi.py#L296-L297

In the original repo, the final action to use is constructed by multiple agents, while the final action is decided by the first agent in this repo, when `agent_obsk` == None.

xihuai18, 2024-09-28:
I make a PR #228 and pass my tests in https://github.com/xihuai18/MaMuJoCo-PettingZoo/blob/main/tests/mamujoco_pettingzoo_v1_test.py, I think these changes make the current implementation perform the same behavior as the original repo.

Kallinteris-Andreas, 2024-09-29:
1. Can you provide an example of the behavior of `agent_obsk=None` when using the original implementation
```
>>> import gymnasium_robotics         
>>> from gymnasium_robotics import mamujoco_v1
>>> env = mamujoco_v1.parallel_env("Ant", "2x4", agent_obsk=None)
>>> env.action_spaces
{'agent_0': Box(-1.0, 1.0, (8,), float32)}
>>> env.observation_spaces
{'agent_0': Box(-inf, inf, (105,), float64)}
```
2. regardless of (1.), what is the reason you want fully observable multi-agent factorizations of the environments, this is equivalent to a single agent environment, with extra steps