Farama-Foundation/Gymnasium-Robotics #239: [Bug Report] Environment toctrees are not being populated in 1.3

Kallinteris-Andreas, 2024-10-11:
In version 1.3 the environment, toctrees are not being populated:
[image]

but in 1.2.4 everything works fine:
[image]

@mgoulao

Kallinteris-Andreas, 2024-10-11:
I have fucked up the registrations of envs during import
I am closing till i fix that, and i will re-open if the issue is not fixed by then

Liang-ZX, 2024-10-15:
Same question

> In version 1.3 the environment, toctrees are not being populated: [...]

Hi, I meet the same problem. Where can I find the doc for v1.2.4?

Kallinteris-Andreas, 2024-10-15:
Bottom left of the docemtation website

Liang-ZX, 2024-10-15:
Thank you so much.

Kallinteris-Andreas, 2024-10-30:
fixed https://github.com/Farama-Foundation/Gymnasium-Robotics/pull/254
will be fixed next release, in the meantime you can use `main` branch (on the bottom left)

closed 2024-11-01 by Kallinteris-Andreas