Farama-Foundation/Gymnasium-Robotics #259: [Bug Report] Collision detection failure in Ant-UMaze

abagaria, 2024-12-01:
**Description of the bug**
While using `AntMaze_UMaze-v5` alongside a pseudocount exploration algorithm, I noticed that the ant can go through the walls in the maze. Initially, this is a rare occurrence, but since I am training an novelty-based exploration algorithm, the agent is able to recreate the issue with greater reliability over time. 

**Code example**
Here is how I am creating the env:

```
env = gym.make(
    'AntMaze_UMaze-v5',
    render_mode='rgb_array',
    max_episode_steps=1000,
    continuing_task=False
)
```
Not that this should be important, but for context, I am using the [TD3](https://arxiv.org/pdf/1802.09477) algorithm and [CFN](https://openreview.net/pdf?id=4RvcXByvnR) for novelty-based intrinsic rewards.

**Versioning**
`gymnasium_robotics`: 1.3.1
`gymnasium`: 1.0.0
`python`: 3.9.20

**Supporting Evidence**

In the attached image, I have plotted the (x, y) coordinates of the ant (according to the states saved in the replay buffer). The color of each point denotes the novelty prediction, but that can be ignored for our purposes. The purple lines show where the walls should be (approximately), and the red circle highlights a trajectory that goes through the wall near the start state and exits near the goal state (which is at `(-4, 4)`).

### Checklist

- [x] I have checked that there is no similar [issue](https://github.com/Farama-Foundation/gymnasium-robotics/issues) in the repo (**required**)
[image]

Kallinteris-Andreas, 2024-12-02:
1) can you provide a video (it might be climbing on top of the wall I am not sure)
2) which `mujoco` version are you using

abagaria, 2024-12-03:
Yup, it looks like we have a flying ant, you gotta love RL :) 

The first attachment is a video showing an example of the ant jumping on top of the maze. The second attachment is a scatter plot which shows that when the ant is in the same `(x, y)` location as the walls, it indeed has a much higher `z-`coordinate than usual, which supports your hypothesis that the ant is jumping on top of the maze.

`mujoco` version: `3.1.6`

https://github.com/user-attachments/assets/25b2ee89-6570-4083-9bb2-357dd362b016
[image]

abagaria, 2024-12-03:
accidentally closed the issue, we still need to figure out how to prevent the ant from jumping on top of the obstacles.

Kallinteris-Andreas, 2024-12-03:
rootz should not be above 1 meter the environment should terminate, are you sure you are checking for `terminated=True`

abagaria, 2024-12-03:
Yes, I am pretty sure i am checking for `terminated=True`. Here is a simple wrapper I am using around the gymnasium_robotics antmaze (it fixes the start, goal states, turns off position noise, and returns observation vectors instead of dictionaries):

```
import numpy as np
import gymnasium_robotics
import gymnasium as gym

from typing import Tuple, Dict

class AntMazeInfoWrapper(gym.Wrapper):
  def __init__(self, env, start_pos=None, goal_pos=None, noise_level=0.0):
    super(AntMazeInfoWrapper, self).__init__(env)
    self.start_pos = start_pos
    self.goal_pos = goal_pos
    self.noise_level = noise_level

    self.unwrapped.position_noise_range = 0.

    if self.goal_pos is not None:
      self.unwrapped.maze._unique_goal_locations = [self.goal_pos]

    if self.start_pos is not None:
      self.unwrapped.maze._unique_reset_locations = [self.start_pos]

  def reset(self) -> Tuple[np.ndarray, Dict]:
    obs, info = self.env.reset()
    
    return self._get_obs(obs), self._modify_info(info, obs)

  def step(self, action) -> Tuple[np.ndarray, float, bool, Dict]:
    obs, reward, terminated, truncated, info = self.env.step(action)

    done = terminated or truncated

    if terminated:
      print(f'Terminated at {obs["achieved_goal"]} with reward {reward}')

    return self._get_obs(obs), reward, done, self._modify_info(info, obs)
  
  def _get_obs(self, obs: Dict):
    pos = obs['achieved_goal']
    return np.concatenate([pos, obs['observation']])
  
  def _modify_info(self, info: Dict, obs: Dict):
    info['start_pos'] = self.start_pos
    info['goal_pos'] = self.goal_pos
    info['pos'] = obs['achieved_goal']
    info['goal'] = obs['desired_goal']
    return info
  

def environment_builder(
  level_name: str = 'AntMaze_UMaze-v5',
  max_episode_steps: int = 800,
  randomize_start_pos: bool = False,
  randomize_goal_pos: bool = False,
  noise_level: float = 0.0
):
  env = gym.make(
    level_name,
    render_mode='rgb_array',
    max_episode_steps=max_episode_steps,
    continuing_task=False)
  
  env = AntMazeInfoWrapper(
    env,
    start_pos=np.array([-4., -4.]) if not randomize_start_pos else None,
    goal_pos=np.array([-4., 4.]) if not randomize_goal_pos else None,
    noise_level=noise_level,
  )

  return env
```

And here is the control loop I used to generate the video I posted in my previous post; as you can see, I am checking for the `done` flag which is set in the wrapper as `done=terminated or truncated`:

```
def rollout(agent, env):
    """Rollout agent in environment."""
    state, _ = env.reset()
    done = False
    total_reward = 0
    state_traj = [state]
    image_traj = [env.render()]

    while not done:
        action = agent.act(state)
        next_state, reward, done, _ = env.step(action)
        total_reward += reward
        state = next_state
        state_traj.append(state)
        image_traj.append(env.render())

    return total_reward, state_traj, image_traj
```

abagaria, 2024-12-05:
Could it be because the AntMaze class is ignoring the terminated flag from the inner mujoco ant env?         `ant_obs, _, _, _, info = self.ant_env.step(action)`

Kallinteris-Andreas, 2024-12-08:
It is indeed doing that
https://github.com/Farama-Foundation/Gymnasium-Robotics/blob/3719d9d79e4f6c5e1ff4f9cbdda821fca880c455/gymnasium_robotics/envs/maze/ant_maze_v5.py#L293-L294

As to why, I don't know it was never an issue before as far as I can tell

The simplest solution would be to create a new file maze_v6 and ant_v6 with the `self.computed_terminated` function changed to take into consideration `info["reward_survive"]`

Kallinteris-Andreas, 2024-12-13:
@abagaria want to try implementing it

abagaria, 2024-12-13:
@Kallinteris-Andreas happy to!

shank-bell, 2026-11-13:
Hi @Kallinteris-Andreas, @abagaria. I'd like to work on this as part of the `maze-v6` effort.

@abagaria, you offered to take this in December 2024. Are you still working on it? If so I'm happy to test or review instead of duplicating work. If not, I'll start.

**Reproduction.** I confirmed the cause on current `main` (gymnasium 1.4.0, mujoco 4.0.1). The walls are 2.0 m tall (box half-height 1.0) and the Ant's `healthy_z_range` is (0.2, 1.0). If I place the torso at z=2.5 (as when standing on a wall) and take one step with a zero action, the inner `ant_env` is unhealthy and returns `terminated=True`. `AntMaze_UMaze-v4` and `-v5`, with `continuing_task` True or False, all return `terminated=False`, because `ant_obs, _, _, _, info = self.ant_env.step(action)` discards the flag.

**Proposal** (new `maze_v6.py` / `ant_maze_v6.py`, so v4/v5 stay reproducible):
- `terminated = inner_ant_terminated or goal_terminated`, using the flag returned by `ant_env.step`.
- Tests: torso above `healthy_z_range` terminates; `terminate_when_unhealthy=False` is respected; goal behaviour is unchanged in both `continuing_task` modes.
- Docs: v6 version history. I also noticed the docs say the goal radius is 0.5 m but the code uses 0.45, so I'll fix that for v6.

**Questions**
1. I'd propagate the inner `terminated` flag rather than check `info["reward_survive"]`. In ant_v5, `reward_survive` is 0 whenever the Ant is unhealthy, even with `terminate_when_unhealthy=False`, so the flag respects that argument. Is that OK?
2. This changes default behaviour. With `continuing_task=True` (the default) episodes will now end when the Ant is unhealthy, whereas the docs say they never terminate. I assume that justifies a new version rather than a patch to v5. Agreed?
3. Naming: `AntMaze-v6` matches the `ant_maze_v6` you proposed. PointMaze is currently v3. Should it become v4 or v6?

I'll keep PRs separate: (1) this fix, (2) goal-switch semantics aligned with #334 / #258, (3) the dense reward from #175.