Farama-Foundation/Gymnasium-Robotics #263: [Bug Report] AntMaze-v4 rendering raises error 'NoneType' object has no attribute 'eglGetCurrentContext'

mengyuest, 2025-01-14:
**Describe the bug**
When I run `AntMaze_Large-v4` environment with EGL setup in a headless server and do some rgb rendering, upon program exit point, an error is raised. `'NoneType' object has no attribute 'eglGetCurrentContext'`

**Code example**
```python
import os
import matplotlib.pyplot as plt
import gymnasium as gym
import gymnasium_robotics
gym.register_envs(gymnasium_robotics)

if __name__ == "__main__":
    os.environ["MUJOCO_GL"] = "egl"
    env = gym.make("AntMaze_Large-v4", render_mode="rgb_array")
    env.reset()
    img = env.render()
    plt.imshow(img)
    plt.savefig("debug_viz.png")
    plt.close()
```

If comment out the `env.render()` line and every line after it, the code can exit normally. Otherwise, it will raise the error:
```bash
Exception ignored in: <function OffScreenViewer.__del__ at 0x788360a7b5b0>
Traceback (most recent call last):
  File "/home/meng/miniconda3/envs/ml/lib/python3.10/site-packages/gymnasium/envs/mujoco/mujoco_rendering.py", line 191, in __del__
  File "/home/meng/miniconda3/envs/ml/lib/python3.10/site-packages/gymnasium/envs/mujoco/mujoco_rendering.py", line 188, in free
  File "/home/meng/miniconda3/envs/ml/lib/python3.10/site-packages/mujoco/egl/__init__.py", line 122, in free
AttributeError: 'NoneType' object has no attribute 'eglGetCurrentContext'
Exception ignored in: <function GLContext.__del__ at 0x7883605495a0>
Traceback (most recent call last):
  File "/home/meng/miniconda3/envs/ml/lib/python3.10/site-packages/mujoco/egl/__init__.py", line 131, in __del__
  File "/home/meng/miniconda3/envs/ml/lib/python3.10/site-packages/mujoco/egl/__init__.py", line 122, in free
AttributeError: 'NoneType' object has no attribute 'eglGetCurrentContext'
```

**System Info**
 * Gymnasium was installed via pip (gymnasium=0.29.1, gymnasium-robotics=1.3.1)
 * Ubuntu 22.04.4 LTS
 * Python 3.10.16

### Checklist

- [x] I have checked that there is no similar [issue](https://github.com/Farama-Foundation/gymnasium-robotics/issues) in the repo (**required**)

Kallinteris-Andreas, 2025-01-14:
1) Does this also happen for gymnasium 1.0?
2) Does this also happen for a simpler environment setters on version 5?

mengyuest, 2025-01-15:
1. The following error happens after installing gymnasium 1.0 in a new conda env. (which I think is the same type of error, stemmed in the __del__ function)
```bash
Exception ignored in: <function OffScreenViewer.__del__ at 0x7c90f2763880>
Traceback (most recent call last):
  File "/home/meng/miniconda3/envs/test_gym/lib/python3.10/site-packages/gymnasium/envs/mujoco/mujoco_rendering.py", line 202, in __del__
  File "/home/meng/miniconda3/envs/test_gym/lib/python3.10/site-packages/gymnasium/envs/mujoco/mujoco_rendering.py", line 199, in free
  File "/home/meng/miniconda3/envs/test_gym/lib/python3.10/site-packages/mujoco/egl/__init__.py", line 124, in free
  File "/home/meng/miniconda3/envs/test_gym/lib/python3.10/site-packages/OpenGL/error.py", line 230, in glCheckError
OpenGL.raw.EGL._errors.EGLError: <exception str() failed>
Exception ignored in: <function GLContext.__del__ at 0x7c90f1df9d80>
Traceback (most recent call last):
  File "/home/meng/miniconda3/envs/test_gym/lib/python3.10/site-packages/mujoco/egl/__init__.py", line 131, in __del__
  File "/home/meng/miniconda3/envs/test_gym/lib/python3.10/site-packages/mujoco/egl/__init__.py", line 124, in free
  File "/home/meng/miniconda3/envs/test_gym/lib/python3.10/site-packages/OpenGL/error.py", line 230, in glCheckError
OpenGL.raw.EGL._errors.EGLError: <exception str() failed>
```

2. A similar error as above happens on ant version 5.

Kallinteris-Andreas, 2025-01-15:
you're not calling the env.close function

mengyuest, 2025-01-15:
You are right. It exits normally now after calling `env.close()` at the end.

closed 2025-01-15 by mengyuest