Farama-Foundation/Gymnasium-Robotics #264: [Bug Report] FetchPickAndPlace-v3 broken

Sebastian-Griesbach, 2025-01-16:
**Describe the bug**
Rendering the FetchPickAndPlace-v3 environment, I see that that robot seems to be misplaced and that the small Box which should be picked up spawns inside the Big Box (table) on the floor.

**Code example**
```
import gymnasium as gym
import gymnasium_robotics

gym.register_envs(gymnasium_robotics)

env = gym.make('FetchPickAndPlaceDense-v3', render_mode='human')

num_episodes = 100
for i in range(num_episodes):
    obs, info = env.reset()
    done = False
    while not done:
        action = env.action_space.sample()
        obs, reward, terminated, truncated, info = env.step(action)
        done = terminated or truncated
```

**System Info**
 installed gymnasium-robotics via pip in a fresh venv
 Ubuntu 24.04.1 LTS
 Python 3.12.3

**Additional context**
[This videos shows the issue.](https://github.com/user-attachments/assets/8458eae4-d2ef-4a50-b28f-74075a7c675a)

### Checklist

- [X] I have checked that there is no similar [issue](https://github.com/Farama-Foundation/gymnasium-robotics/issues) in the repo (**required**)

kuds, 2025-01-16:
Hey @Sebastian-Griesbach, this is a known issue in the current release version of Gymnasium Robotics and was fixed in PR #256. 

If you do a pip install from the source and use FetchPickAndPlaceDense-v4, it should work.

`pip install git+https://github.com/Farama-Foundation/Gymnasium-Robotics.git`

Sebastian-Griesbach, 2025-01-17:
Thank you for letting me know. This works :)