Farama-Foundation/Gymnasium-Robotics #265: [Bug Report] Pointmaze (and maybe other environments) broken with Gymnasium v1.1.0

mkschleg, 2025-02-28:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

**Describe the bug**

Mujoco environments are incompatible with new version of gymnasium. Need to update the render_modes metadata entry.

**Code example**
In a fresh python environment:
```
pip install gymnasium==1.1.0 gymnasium_robotics
```
In the repl:
```
>>> import gymnasium
>>> import gymnasium_robotics
>>> env = gymnasium.make("PointMaze_UMaze-v3")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/matt/.pyenv/versions/test-gymnasium/lib/python3.10/site-packages/gymnasium/envs/registration.py", line 742, in make
    env = env_creator(**env_spec_kwargs)
  File "/Users/matt/.pyenv/versions/test-gymnasium/lib/python3.10/site-packages/gymnasium_robotics/envs/maze/point_maze.py", line 342, in __init__
    self.point_env = PointEnv(
  File "/Users/matt/.pyenv/versions/test-gymnasium/lib/python3.10/site-packages/gymnasium_robotics/envs/maze/point.py", line 42, in __init__
    super().__init__(
  File "/Users/matt/.pyenv/versions/test-gymnasium/lib/python3.10/site-packages/gymnasium/envs/mujoco/mujoco_env.py", line 85, in __init__
    assert self.metadata["render_modes"] == [
AssertionError: ['human', 'rgb_array', 'depth_array']
```

**System Info**
Describe the characteristic of your environment:
- Installed through pip
- MacOS
- Python version 3.10.13

### Checklist

- [x] I have checked that there is no similar [issue](https://github.com/Farama-Foundation/gymnasium-robotics/issues) in the repo (**required**)

shanpenghui, 2025-03-01:
I try but failed, so have to rollback the version of gymnasium.
```
pip uninstall -y gymnasium
pip install gymnasium==1.0.0
```

Kallinteris-Andreas, 2025-03-01:
This propably has to do with the addition of render_mode=`rgbd_tuple`

closed 2025-03-06 by Kallinteris-Andreas