Farama-Foundation/Gymnasium-Robotics #268: [Bug Report] Bug title

ekkooee7, 2025-04-16:
​​First, thank you for developing and maintaining this fantastic robotics simulation environment!​​ The FetchPickAndPlace task has been incredibly valuable for our research. 

In the FetchPickAndPlace-v3 environment from gymnasium-robotics, the block (object) sometimes spawns ​​under the table​​ instead of on top of it. This makes the task impossible since the robot cannot reach objects below the table surface.

ekkooee7, 2025-04-17:
I use mujoco210. I do not set a seed in specific, the block always locates at the bottom inside the table where the robot can't reach. I tried to find where to set the initial position of block, but seems that it's set by the xml file which is not included in the repo

Mgineer117, 2025-05-30:
Did you ever figure it out? I am having the same problem with using mujoco 210.