Farama-Foundation/Gymnasium-Robotics #269: [Bug Report] Fetch robot misplaced

lubiluk, 2025-04-18:
In Fetch environments (FetchReach-v3, FetchPush-v3...) Fetch robot appears nowhere near the table, so solving the environment is impossible.

**Code example**
```
import gymnasium as gym
import gymnasium_robotics

gym.register_envs(gymnasium_robotics)
env = gym.make("FetchReach-v3", render_mode="human")
env.reset()
```

**System Info**
gymnasium-robotics==1.3.1
 * installed with pip
 * Tested on Python 3.12 on MacOS, and Python 3.11 on Linux

kuds, 2025-04-23:
Hey @lubiluk , this is a known issue (#251 &  #264) in the current release version (1.3.1) of Gymnasium Robotics and was fixed in PR https://github.com/Farama-Foundation/Gymnasium-Robotics/pull/256.

If you do a pip install from source and use FetchPickAndPlaceDense-v4, it should work.

`pip install git+https://github.com/Farama-Foundation/Gymnasium-Robotics.git`

lubiluk, 2025-04-24:
Right, duplicate. Sorry.

closed 2025-04-24 by lubiluk